ikwzm/udmabuf #128: using 64-bit high addr cma failed

Jeason-Hu, 2025-01-14:
Dear @ikwzm,

I have encountered some cma address problems, could you please give me a hand?
when I use 64-bit low addr, it comes ok:

```DeviceTree
reserved-memory {  
		#address-cells= <2>;  
		#size-cells= <2>;  
		ranges;  
		run_buffer: run_buffer@0 {  
			compatible = "shared-dma-pool";  
			reusable;  
			reg = <0x00000000 0x61000000 0x00000000 0x04000000>;
			label = "run_buffer";  
		}
};
udmabuf@0x00 {
	compatible = "ikwzm,u-dma-buf";
	device-name = "runlength";
	size = <0x0 0x04000000>;
	memory-region = <&run_buffer>;
};
```

\#insmod /mnt/u-dma-buf.ko   
[   36.083899] u_dma_buf: module is from the staging directory, the quality is unknown, you have been warned.  
[   36.094751] u-dma-buf udmabuf@0x00: assigned reserved memory node run_buffer@0  
[   36.118523] u-dma-buf runlength: driver version = 5.0.3  
[   36.123756] u-dma-buf runlength: major number   = 239  
[   36.128809] u-dma-buf runlength: minor number   = 0  
[   36.133682] u-dma-buf runlength: phys address   = 0x0000000061000000  
[   36.140029] u-dma-buf runlength: buffer size    = 67108864  
[   36.145505] u-dma-buf udmabuf@0x00: driver installed.  

but when I change to high addr, it failed.

```DeviceTree
reserved-memory {  
		#address-cells= <2>;
		#size-cells= <2>;
		ranges;
		run_buffer: run_buffer@0 {
			compatible = "shared-dma-pool";
			reusable;
			reg = <0x00000008 0x61000000 0x00000000 0x04000000>;
			label = "run_buffer";
		}
};
udmabuf@0x00 {
	compatible = "ikwzm,u-dma-buf";
	device-name = "runlength";
	size = <0x0 0x04000000>;
	memory-region = <&run_buffer>;
};
```
\#insmod /mnt/u-dma-buf.ko  
[   15.908861] u_dma_buf: module is from the staging directory, the quality is unknown, you have been warned.  
[   15.919771] u-dma-buf udmabuf@0x00: assigned reserved memory node run_buffer@0  
[   15.936419] ------------[ cut here ]------------  
[   15.941034] WARNING: CPU: 2 PID: 211 at mm/page_alloc.c:5356 __alloc_pages+0x178/0xb7c  
[   15.948949] Modules linked in: u_dma_buf(C+)  
[   15.953212] CPU: 2 PID: 211 Comm: insmod Tainted: G         C        5.15.0-g5a22cdd09df7-dirty #11  
[   15.962246] Hardware name: ZynqMP (DT)  
[   15.967021] pstate: 20000005 (nzCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)  
[   15.973973] pc : __alloc_pages+0x178/0xb7c  
[   15.978061] lr : __dma_direct_alloc_pages+0x17c/0x1f0  
[   15.983105] sp : ffffffc012b6b630  
[   15.986403] x29: ffffffc012b6b630 x28: 0000000000000013 x27: 0000000000000100  
[   15.993529] x26: ffffffc008cb7500 x25: ffffffc0100cf7a0 x24: ffffffc0100aa0ac  
[   16.000656] x23: ffffff88003fd410 x22: 0000000000000000 x21: 000000000000000e  
[   16.007782] x20: 0000000000000cc4 x19: 0000000004000000 x18: ffffffc010fe2000  
[   16.014908] x17: ffffff8800001008 x16: fffffffe1d617fd0 x15: 0000000000000001  
[   16.022035] x14: fffffffe00000000 x13: 6db6db6db6db6db7 x12: ffffffc86e6cd000  
[   16.029161] x11: ffffff887f7accb0 x10: ffffffc86e6cd000 x9 : ffffffc011302a00  
[   16.036288] x8 : fffffffe1d617fd0 x7 : 0000000000000000 x6 : 0000000000003fc0  
[   16.043414] x5 : ffffff8800257000 x4 : 0000000000000032 x3 : 0000000000000000  
[   16.050540] x2 : ffffff8800370600 x1 : 0000000000000000 x0 : 0000000000000cc4  
[   16.057667] Call trace:  
[   16.060098]  __alloc_pages+0x178/0xb7c  
[   16.063838]  __dma_direct_alloc_pages+0x17c/0x1f0  
[   16.068534]  dma_direct_alloc+0x70/0x380  
[   16.072449]  dma_alloc_attrs+0x7c/0xe4  
[   16.076190]  udmabuf_object_setup+0x44/0x1fc [u_dma_buf]  
[   16.081511]  udmabuf_platform_driver_probe+0x378/0x6bc [u_dma_buf]  
[   16.087691]  platform_probe+0x68/0xe0  
[   16.091346]  really_probe.part.0+0x9c/0x30c  
[   16.095520]  __driver_probe_device+0x98/0x144  
[   16.099869]  driver_probe_device+0x44/0x11c  
[   16.104044]  __driver_attach+0xf8/0x190  
[   16.107872]  bus_for_each_dev+0x70/0xd0  
[   16.111700]  driver_attach+0x24/0x30  
[   16.115268]  bus_add_driver+0x104/0x1ec  
[   16.119096]  driver_register+0x78/0x130  
[   16.122923]  __platform_driver_register+0x28/0x34  
[   16.127619]  u_dma_buf_init+0x20c/0x1000 [u_dma_buf]  
[   16.132585]  do_one_initcall+0x50/0x1b0  
[   16.136412]  do_init_module+0x54/0x250  
[   16.140153]  load_module+0x2008/0x2580  
[   16.143895]  __do_sys_finit_module+0xa4/0xf0  
[   16.148156]  __arm64_sys_finit_module+0x24/0x30  
[   16.152679]  invoke_syscall+0x54/0x124  
[   16.156420]  el0_svc_common.constprop.0+0x44/0xfc  
[   16.161116]  do_el0_svc+0x48/0xb0  
[   16.164423]  el0_svc+0x28/0x80  
[   16.167470]  el0t_64_sync_handler+0xa4/0x130  
[   16.171731]  el0t_64_sync+0x1a0/0x1a4  
[   16.175386] ---[ end trace a8036ad8f7d21654 ]---  
[   16.180021] u-dma-buf runlength: dma_alloc_coherent(size=67108864) failed. return(0)  
[   16.187765] u-dma-buf udmabuf@0x00: object setup failed. return=-12  
[   16.194211] u-dma-buf udmabuf@0x00: driver probe failed. return=-12  
[   16.200486] u-dma-buf: probe of udmabuf@0x00 failed with error -12

ikwzm, 2025-01-14:
Thanks for the issue.

Could you try adding the dma-mask property to the u-dma-buf node as shown next?

```dts
udmabuf@0x00 {
	compatible = "ikwzm,u-dma-buf";
	device-name = "runlength";
	size = <0x0 0x04000000>;
	dma-mask = <40>;
	memory-region = <&run_buffer>;
};
```

The value you set for the dma-mask property depends on the hardware you are using. If your system's physical address is 40 bits, set it to 40.

Jeason-Hu, 2025-01-14:
> Thanks for the issue.
> 
> Could you try adding the dma-mask property to the u-dma-buf node as shown next?
> 
> ```
> udmabuf@0x00 {
> 	compatible = "ikwzm,u-dma-buf";
> 	device-name = "runlength";
> 	size = <0x0 0x04000000>;
> 	dma-mask = <40>;
> 	memory-region = <&run_buffer>;
> };
> ```
> 
> The value you set for the dma-mask property depends on the hardware you are using. If your system's physical address is 40 bits, set it to 40.

It works. Thank you very much.