ikwzm/udmabuf #131: Failed to create large buffer : vmap allocation for size XXX  failed

doonny, 2025-02-06:
Hi, I am trying to create large buffers larger than 130M, but the driver failed with the following errors:

[image]

I have set CMA=512M as follow, still can not work:

[image]

I am using petalinux v2021.2

[image]

How could I solve this problem ?  Thanks

ikwzm, 2025-02-06:
Thank you for the issue.

I assume you are using MicroBraze, but I am not familiar with it. 
Perhaps it is a limitation of Linux for MicroBraze.
Similar symptoms are described in issue #33.

ikwzm, 2025-02-06:
I asked ChatGPT asked “How do I change the vmalloc space in linux kernel with arm32?” and the following answer was given.

<html>
<body>
<html><head></head><body><h2><strong>How to Increase <code inline="">vmalloc</code> Space in Linux Kernel on ARM32</strong></h2>
<p>On <strong>ARM32 Linux</strong>, the <strong><code inline="">vmalloc</code> space</strong> is limited due to the <strong>32-bit virtual address space (4GB)</strong>, which is shared between <strong>kernel space and user space</strong>.<br>
By default, the <code inline="">vmalloc</code> space is <strong>small</strong> because the kernel reserves most of the virtual address space for <strong>direct-mapped memory</strong> and <strong>other kernel structures</strong>.<br>
If you need to <strong>increase the <code inline="">vmalloc</code> space</strong>, you can do so by <strong>modifying the kernel configuration</strong> or <strong>changing the kernel boot parameters</strong>.</p>
<hr>
<h2><strong>🔹 1. Understanding the <code inline="">vmalloc</code> Address Space on ARM32</strong></h2>
<p>On <strong>ARM 32-bit</strong>, the virtual memory layout is typically structured as follows:</p>
<pre><code>0x00000000 - 0xBFFFFFFF  User space (3GB, typical)
0xC0000000 - 0xEFFFFFFF  Kernel direct-mapped memory (768MB)
0xF0000000 - 0xFFBFFFFF  vmalloc, ioremap, and highmem mappings
0xFFC00000 - 0xFFFFFFFF  Kernel reserved area (4MB)
</code></pre>
<ul>
<li>The <code inline="">vmalloc</code> area is usually around <strong>240MB–256MB</strong> (<code inline="">0xF0000000</code> to <code inline="">0xFFBFFFFF</code>).</li>
<li>If you need <strong>more vmalloc space</strong>, you have to <strong>reduce the kernel direct-mapped memory region</strong>.</li>
</ul>
<hr>
<h2><strong>🔹 2. Method 1: Adjusting <code inline="">CONFIG_VMSPLIT</code></strong></h2>
<p>The <strong>easiest way to increase <code inline="">vmalloc</code> space</strong> is to change the <strong><code inline="">CONFIG_VMSPLIT</code></strong> setting in the kernel.</p>
<ol>
<li>
<p><strong>Open kernel configuration menu:</strong></p>
<pre><code class="language-bash">make menuconfig
</code></pre>
</li>
<li>
<p><strong>Go to:</strong></p>
<pre><code>Kernel Features  ---&gt;  
  Memory split  ---&gt;
</code></pre>
</li>
<li>
<p><strong>Choose a different <code inline="">VMSPLIT</code> option:</strong></p>
<ul>
<li><code inline="">CONFIG_VMSPLIT_3G</code> (Default) → <strong>User: 3GB, Kernel: 1GB</strong> (vmalloc: ~240MB)</li>
<li><code inline="">CONFIG_VMSPLIT_2G</code> → <strong>User: 2GB, Kernel: 2GB</strong> (vmalloc: ~768MB)</li>
<li><code inline="">CONFIG_VMSPLIT_1G</code> → <strong>User: 1GB, Kernel: 3GB</strong> (vmalloc: ~1.5GB)</li>
</ul>
</li>
<li>
<p><strong>Save and rebuild the kernel:</strong></p>
<pre><code class="language-bash">make -j$(nproc)
sudo make modules_install install
sudo reboot
</code></pre>
</li>
</ol>
<p>✅ <strong>Using <code inline="">VMSPLIT_2G</code> or <code inline="">VMSPLIT_1G</code> increases <code inline="">vmalloc</code> space significantly!</strong><br>
🚨 <strong>However, reducing user space may break some applications that require a large user address space.</strong></p>
<hr>
<h2><strong>🔹 3. Method 2: Increase <code inline="">vmalloc</code> Space via Kernel Boot Parameters</strong></h2>
<p>You can <strong>increase <code inline="">vmalloc</code> space without recompiling the kernel</strong> by modifying the <strong>boot parameters</strong>.</p>
<h3><strong>Steps to Modify Boot Parameters</strong></h3>
<ol>
<li>Edit the bootloader configuration (<code inline="">u-boot</code>, <code inline="">grub</code>, etc.).</li>
<li>Add the <code inline="">vmalloc=</code> parameter to the kernel boot arguments.</li>
</ol>
<p><strong>Example: Increasing <code inline="">vmalloc</code> to 512MB</strong></p>
<pre><code class="language-bash">vmalloc=512M
</code></pre>
<p>For <strong>U-Boot:</strong></p>
<pre><code class="language-bash">setenv bootargs ${bootargs} vmalloc=512M
saveenv
</code></pre>
<p>For <strong>GRUB (if applicable):</strong></p>
<pre><code class="language-bash">sudo nano /etc/default/grub
</code></pre>
<p>Modify:</p>
<pre><code class="language-bash">GRUB_CMDLINE_LINUX_DEFAULT="quiet splash vmalloc=512M"
</code></pre>
<p>Then update:</p>
<pre><code class="language-bash">sudo update-grub
sudo reboot
</code></pre>
<p>✅ <strong>Using <code inline="">vmalloc=512M</code> increases <code inline="">vmalloc</code> space dynamically.</strong><br>
🚨 <strong>However, this method may not work on some ARM32 systems if <code inline="">VMSPLIT</code> is too restrictive.</strong></p>
<hr>
<h2><strong>🔹 4. Checking the Current <code inline="">vmalloc</code> Space</strong></h2>
<p>After rebooting, verify the new <code inline="">vmalloc</code> allocation with:</p>
<pre><code class="language-bash">cat /proc/meminfo | grep Vmalloc
</code></pre>
<p>Example output:</p>
<pre><code class="language-plaintext">VmallocTotal:   536870912 kB  # 512MB vmalloc
VmallocUsed:      120000 kB
VmallocChunk:    400000 kB
</code></pre>
<hr>
<h2><strong>🔹 5. Summary</strong></h2>

Method | How it Works | Pros | Cons
-- | -- | -- | --
Modify CONFIG_VMSPLIT | Adjusts kernel/user space split | More reliable, large increase | Requires kernel recompilation
Set vmalloc= boot parameter | Expands vmalloc dynamically | No need to recompile kernel | Limited by VMSPLIT setting

<p>✅ <strong>If you need a <strong>permanent</strong> increase, change <code inline="">CONFIG_VMSPLIT</code> and recompile the kernel.</strong><br>
✅ <strong>If you need a <strong>temporary</strong> increase, use the <code inline="">vmalloc=</code> boot parameter.</strong></p>
<p>🚀 <strong>Increasing <code inline="">vmalloc</code> space allows larger kernel allocations, improving performance for drivers and large memory mappings!</strong></p></body></html>
</body>
</html>

doonny, 2025-02-07:
@ikwzm Thanks for you quick reply.  
After enlarging the vmalloc space, this issue was solved. 
One more question, there is any performance penalties when using very large buffers ? for example >200MB ?  or do you recommend always try to partition large buffers to small ones ?

For anyone that might faced the same issue, you could refer to [https://adaptivesupport.amd.com/s/question/0D52E00006hpiaeSAA/how-to-increase-size-of-vmalloc-for-petalinux-on-microblaze?language=zh_CN](url) for detailed information on how to change the vmalloc setting in petalinux.

ikwzm, 2025-02-07:
Thank you for the useful information.

> One more question, there is any performance penalties when using very large buffers ? for example >200MB ? or do you recommend always try to partition large buffers to small ones ?

It is very difficult for me to give you a general answer because the situation varies depending on your system.
If there is any advice I can give you, it relates to cache operations.
For example, cache flushes and invalidates are slower the larger the target area, so if you have to divide an area into multiple buffers, please be careful about the size of the cache operation.

tgrygorii, 2025-07-29:
The issue with VMSPLIT_2G and vmalloc= is that it reduces the available memory for standard malloc().

For instance, with 3G of physical RAM on an arm32 CPU, VMSPLIT_2G allocates 2G of virtual space for user applications. If the u-dma-buf is 1G, the application can mmap() 1G on /dev/udmabuf0 and allocate a maximum of 1G (often less) with malloc().

@ikwzm

Is it possible to allocate or reserve memory for the u-dma-buf buffer without using vmalloc(), if the buffer is intended solely for user space?