ikwzm/udmabuf #137: Writing PL Data on a MPSoC to an SSD, quirk-mmap-on question

smartheit, 2025-08-05:
Hello @ikwzm,
I am currently writing an application note showing how to write PL data to and SSD connected via PCIe.
As a buffer manager I am using your driver, thank you very much for this work.

This is the udmabuf config:

> 	dmabuf@0x00 {
> 		compatible = "ikwzm,u-dma-buf";
> 		device-name = "udmabuf0";
> 		minor-number = <0>;
> 		quirk-mmap-on;
> 		size = <0x10000000>;
> 	};

For writing to the ssd I am using io_uring ( https://en.wikipedia.org/wiki/Io_uring).

I thought with **_quirk-mmap-on_** a mmaped udma buffer behaves the same as a buffer allocated with malloc. This code snippet:

```
    // 6. Wait for completion
    io_uring_cqe* cqe;
    if (io_uring_wait_cqe(&ring, &cqe) < 0) {
        perror("io_uring_wait_cqe");
        io_uring_queue_exit(&ring);
        close(fd);
    }

    auto end = std::chrono::steady_clock::now();
    std::chrono::duration<double, std::milli> duration_ms = end - start;

    if (cqe->res < 0) {
        fprintf(stderr, "Async write failed: %s\n", strerror(-cqe->res));
```
will fail with an error (-5) Input/Output error. 

When I use`std::memcpy(buffer, dmaBufLow, C_HALF_DMA_BUF_SIZE);` to copy data from the udma buffer to the malloc buffer the ssd writes do work.
(but the performance drops and memory bandwidth is wasted)

Do you have an idea if it is possible to use udmabuf the same was as a normal user space buffer?

Thanks
Marco

smartheit, 2025-08-06:
I attached the whole project, it is in a test and tinker state (concept):
[dma_server.zip](https://github.com/user-attachments/files/21611085/dma_server.zip)
The relevant files are main.cpp and thdDiscIo.cpp.
In main.cpp the uio and udmabuf are initialized and the interrupts of the uio device are handled (~10 Hz as the udmabuf is 128MB). In thdDiscIo.cpp line I am currently copying the udmabuffer to a malloc buffer and the goal is to avoid that.
If this would (direct udmabuf) work 400MByte/s SSD writes can be achieved with little CPU load which is close to the maximum PCIe x1 bandwith.

ikwzm, 2025-08-06:
I'm sorry, but I don't understand the reason why an "IO Error" occurs.
In the case of a "Bad Address" error, it can sometimes be avoided by specifying ```quirk-mmap-page``` in the device tree for u-dma-buf.

Currently under development, but a sample using io_uring with u-dma-buf is shown at the following URL.

  * https://github.com/ikwzm/u-dma-buf-test/tree/develop

The device tree uses the following.

  * https://github.com/ikwzm/u-dma-buf-test/blob/develop/dts/udmabuf0-page-dma-coherent.dts

```dts:https://github.com/ikwzm/u-dma-buf-test/blob/develop/dts/udmabuf0-page-dma-coherent.dts
/dts-v1/; /plugin/;
/ {
	fragment@0 {
		target-path = "/amba_pl@0";
		#address-cells = <2>;
		#size-cells = <2>;

		__overlay__ {
			#address-cells = <2>;
			#size-cells = <2>;

			udmabuf0 {
				compatible  = "ikwzm,u-dma-buf";
				device-name = "udmabuf0";
				size = <0x02000000>;
				dma-coherent;
				quirk-mmap-page;
	                };
                };
	} ;
} ;
```

u-dma-buf-uring-test.c is located at the following URL.

  * https://github.com/ikwzm/u-dma-buf-test/blob/develop/u-dma-buf-uring-test.c

When I ran u-dma-buf-uring-test, the following happened.

When running u-dma-buf-uring-test on KV260 + Debian12 + linux kernel 6.12.27, the following occurred.

```console
fpga@debian-fpga:~/work/u-dma-buf-test$ sudo ./u-dma-buf-uring-test
device=udmabuf0
driver_version=5.3.0
dma_coherent=1
quirk_mmap=page
size=33554432
write_buf_test(size=33554432, O_DIRECT=0)
time = 1.840972 sec
time = 1.904498 sec
time = 1.984143 sec
time = 1.864012 sec
time = 1.928946 sec
time = 1.968296 sec
time = 1.877946 sec
time = 1.931692 sec
time = 1.862262 sec
time = 2.007904 sec
write_buf_test(size=33554432, O_DIRECT=1)
time = 1.762077 sec
time = 1.742016 sec
time = 1.777801 sec
time = 1.720855 sec
time = 1.719468 sec
time = 1.715296 sec
time = 1.721440 sec
time = 1.696683 sec
time = 1.726461 sec
time = 1.730213 sec
```

Your situation may differ, but please use this as a reference.