ikwzm/udmabuf #140: Device Tree building does not pick up declared reserved-memory entry and caching seems to be disabled for memory-mapped IO

MarioTrams2025, 2025-09-16:
Hello,

I understand the u-dma-buf driver is essentially targeted on ZYNQ. I'm trying to apply it to another micro controller containing the Cortex-A53 (TexasInstruments AM6442). Because the u-dma-buf driver is also supporting Cortex-A53 in ZYNQ, I believe this should not be an issue.

I did chose to integrate the u-dma-buf driver into the kernel sources as it has been described here.  The  Linux kernel version is 6.12.35 - probably patched by TexasInstruments at some places - it originates from their current SDK version 11.01.05.03). 

What I wanted to do is actually NOT getting cachable access to some (reserved) part of main memory, but to a dedicated region in physical memory. With an mmap() of /dev/mem I'm getting access to such a region, but without active caching.

I did follow the u-dma-buf documentation and did add an entry into the "reserved-memory" section of the device tree configuration. 

To quote the relevant part of the device tree:

```
                 fpga_fast_fifo: fpga_fast_fifo-memory@53000000 {
                         compatible = "shared-dma-pool";
                         reusable;
                         reg = <0x00 0x053000000 0x00 0x00020000>;
                         label = "image_buf0";
                 };
         };
         udmabuf@0 {
                 compatible = "ikwzm,u-dma-buf";
                 devicename = "udmabuf0";
                 size = <0x00020000>;
                 memory-region = <&fpga_fast_fifo>;
         };
 
```

Now, when the device tree is being built, I'm receiving the following warning:
arch/arm64/boot/dts/ti/k3-am642-evm.dts:157.12-172.4: Warning (unit_address_vs_reg): /udmabuf@0: node has a unit name, but no reg or ranges property

When I load the module later on via insmod as specified in the documentation (note: I have to rmmod the module beforehand since it is being automatically loaded without parameters during boot), I'm receiving the following output:
[   83.046612] u_dma_buf: module is from the staging directory, the quality is unknown, you have been warned.
[   83.058816] u-dma-buf udmabuf@0: of_reserved_mem_device_init failed. return=-22
[   83.066840] u-dma-buf udmabuf@0: driver probe failed. return=-22
[   83.073161] u-dma-buf udmabuf@0: probe with driver u-dma-buf failed with error -22
[   83.083660] u-dma-buf udmabuf0: driver version = 5.3.0
[   83.089051] u-dma-buf udmabuf0: major number   = 235
[   83.094125] u-dma-buf udmabuf0: minor number   = 0
[   83.099041] u-dma-buf udmabuf0: phys address   = 0x00000000fbc20000
[   83.105537] u-dma-buf udmabuf0: buffer size    = 131072
[   83.113935] u-dma-buf u-dma-buf.0: driver installed.

I can't really judge these first three error messages. The reported physical address of 0xfbc20000 is not the expected one (0x53000000) but some address in the upper part of the regular main memory. I don't know from where the system is taking that address. Anyway, I tried to mmap() and access this buffer, and this seems to work well. I can also either specify O_SYNC (or not) during opening the device file and based on performance figures I can clearly see whether the cache is present or not. So this seems just fine - only the wrong physical address.

I played a lot with the device tree settings in order to get rid of this warning and eventually get the right physical address into the driver. I had no success.
So I tried to hard-code the address into the driver with a dirty hack. I did this by manipulating udmabuf_platform_device_probe(). Almost at the end of that function just prior to unlocking the mutex I added:
obj->phys_addr = 0x053000000;

Indeed, that seems to work. I'm receiving the proper physical address in the debugging output. I can even mmap() that area, access it, and transactions get forwarded to the hardware. However, the caching seems turned off. That is, no matter whether O_SYNC is specified or not, I'm not seeing a difference and the overall performance is rather low. Interestingly, I'm seeing a write performance way better than via mmaping of /dev/mem - which is probably due to some different write combining policy. The read performance is even a little bit worse compared to /dev/mem.

So the primary question is: Why is there no caching?

One suspicion I have: Linux knows where there is regular memory and assumes that only this memory can be cached. The rest is memory mapped IO which is uncachable (which is certainly correct - unless software is not taking control over cache coherency). Probably u-dma-buf inherits these settings and doesn't touch them. Can this be confirmed? If yes, are there some work-arounds?

There is this memory-node within the linux device tree settings. In my particular case there is just specified  the regular main memory. I did add a second node specifying the window I need. However, this is raising other issues. In particular there is a kernel panic during boot. This is just a guess of mine, but it might be that Linux is accessing this special region  in order to initialize it or whatsoever. But the problem is that the hardware needs to be set up first by a special driver that gets loaded later in the boot process. 

Thanks in advance for any hints that could improve the situation.

With best regards,
Mario

ikwzm, 2025-09-16:
Thank you for the issue.

The Linux CMA region has alignment constraints.
For MPSoC, the address and size that can be specified in the regs property of reserved-memory must be in units of 0x00400000 (4MiB).

ikwzm, 2025-09-19:
> Seems I could solve that problem...

First of all, congratulations on getting one problem resolved.

> So this defines 2GiB of main memory starting at 0x080000000. The physical base address of the memory-mapped IO window I wanted to map is 0x053000000 - which is below 0x080000000 resp. outside the declared memory area.

Could it be that the memory region starting at 0x5300000 is memory provided on the PL side? If so, issue #45 might be helpful.

MarioTrams2025, 2025-09-19:
> Could it be that the memory region starting at 0x5300000 is memory provided on the PL side? If so, issue https://github.com/ikwzm/udmabuf/issues/45 might be helpful.

Indeed, if I would have read through all of the historic issues I would have found the proper hint in #45 as well. Though, I did not have to deal with that dma-mask setting.

The term "PL side" refers more to the ZYNQ FPGA terminology I believe, but from the perspective of the processor it is the same in my case. So in the end it is just some memory-mapped window of an IO unit that I'm using in order to attach memory within an external FPGA to the processor.

There is also another dma_buf driver in Linux I did experiment with before I came across u-dma-buf. I was failing there as well, but I had also not yet added this additional declaration of the memory area. Perhaps I should try this again.

Anyway, thanks for your support on that matter!